budulinek/arduino-modbus-rtu-tcp-gateway #3: warning compilation

jmcastillejo, 2021-04-12:
Warning compiler for mega2560: 

arduino-modbus-rtu-tcp-gateway\04-webserver.ino:288:20: warning: invalid conversion from 'byte {aka unsigned char}' to 'action_type' [-fpermissive]
           action = (byte)atoi(strtok_r(NULL, "=", &sav2));

budulinek, 2021-04-12:
Yes, I know. 

In order to minimize the code size (so that it can fit Nano), I did a dirty trick with atoi() while parsing the "Tools" form values (buttons):

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/51595a10fe23155318b2775a49f6ef0d00921b43/arduino-modbus-rtu-tcp-gateway/04-webserver.ino#L288

I think it is save to ignore the warning, because the form values are fully under our control and can only have values 1, 2 and 3.
see https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/51595a10fe23155318b2775a49f6ef0d00921b43/arduino-modbus-rtu-tcp-gateway/05-pages.ino#L547

Also, see comments in code:
https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/51595a10fe23155318b2775a49f6ef0d00921b43/arduino-modbus-rtu-tcp-gateway/04-webserver.ino#L24

budulinek, 2021-04-12:
That did the trick :-)

closed 2021-04-12 by budulinek