budulinek/arduino-modbus-rtu-tcp-gateway #5: timeout error

jmcastillejo, 2021-04-13:
I have managed to make the project work on an arduino mega 2560, but the master simulator throws me timeout error in alternate reads, one ok and another reading timeout error and so on. that could be happening?

great job

budulinek, 2021-04-13:
Which master simulator do you use?

Do you have errors on both Modbus TCP and Modbus UDP? 

What is your connection timeout on master? What is your timeout on Arduino?

Thanks.

jmcastillejo, 2021-04-14:
I use de radzio master simulator http://en.radzio.dxp.pl/modbus-master-simulator/

[image]
[image]

This softaware not support modbus udp. In arduino timeout set to 1000ms, although with other configurations it still fails the same

budulinek, 2021-04-14:
Thanks. Yes, I can reproduce the error. Modbus UDP is OK, so the problem is in the way I use TCP sockets...

jmcastillejo, 2021-04-19:
Do you have any solution?

tdcbyte, 2021-05-24:
I confirm the issue, also tested with arduino mega. UDP works fine, when using TCP, after the first request, the response is good, but second request is timed out en not send to the serial slave. The the third request is ok, the forth is not answered and so on...
I hope this can be fixed, still this project is really nice!

fumpel, 2021-06-03:
I think, I'm running into the same problem... I'm using the QModMaster. Behaviour is little different though: After opening a TCP connection the first request works fine, all subsequent requests don't. Reopening the connection again gives one working request.

fumpel, 2021-06-04:
Had a look at the code...
Could it be, that this line:
[https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/16246c5eff7095fcb870da9db0f55f82c50b81ea/arduino-modbus-rtu-tcp-gateway/02-modbus-tcp.ino#L136](client.stop())
is the issue? Isn't this closing the TCP Connection after every received packet? Then the TCP Modbus Master would have to reestablish the connection after every packet... simple Modbus Masters won't do that.

budulinek, 2021-06-05:
Essentially there are two ways how to handle sockets:

First aproach:
https://www.arduino.cc/en/Tutorial/LibraryExamples/WebServer
`EthernetClient client = server.available();`

Second approach:
https://www.arduino.cc/en/Tutorial/LibraryExamples/AdvancedChatServer
you declare before setup:
`EthernetClient clients[8];`
and the inside loop (or some nested function):
```
EthernetClient newClient = server.accept();
  if (newClient) {
    for (byte i=0; i < 8; i++) {
      if (!clients[i]) {
        clients[i] = newClient;
        break;
      }
    }
  }
```

Originally, I used the second approach and everything worked fine. Then I switched to the first approach (to save some code) and it broke the TCP connections. Sockets get closed and I do not even know whether it is because of my code or the library.

I was busy at work so did not have time to look at it.
@fumpel I do not think removing this client.stop(); solves the issue.

budulinek, 2021-06-06:
@fumpel
Oh man, you were right. Removing all instances of  client.stop() solved the issue. 

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/commit/574c3cc78f476bb6d9bc1779059331fc5668aac1

We do not need client.stop(), because the server.available() function should take care of this (it should close connection if the remote hos closes the connection).

If the problem reappears, let me know.

closed 2021-06-06 by budulinek

SuriyaStar123, 2021-12-20:
[image]
  
How to get off this error and write numbers in 0 and check with modscan