budulinek/arduino-modbus-rtu-tcp-gateway #6: Run this Gateway on a ESP8266

turti83, 2021-05-03:
I am new on github and dont know how to ask a questions like this way. Is this the right way?
This project is exactly what i was looking for. However, I would like to run it on an ESP8266. Is it possible without a lot of changes, or does it take a lot of effort to run it on the esp? 
certainly the following line must be inserted:
#include <ESP8266WiFi.h>
and removed this:
#include <utility/w5100.h>

budulinek, 2021-06-06:
Sorry, but I do have no experience with ESP8266.
There is some code which is specific to Arduino (see https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway#portability). So some modifications are needed.
Feel free to fork my repository and modify the code for ESP8266.

harihanv, 2021-07-14:
https://github.com/harihanv/esp32-modbus-gateway, trying to port on esp32

ayaser, 2021-12-27:
Nice to meet you. first congratulations for your contribution. I have several doubts, one of them is the same as the one made by Turti83, how to use these codes in ESP8266, and I clearly saw that your answer is that you have not developed it on this platform.
The other question I have is that there are several ino files in the folder, and the question I have is whether each of these can be loaded on the platform that one would like independently. That is, if I want the Slave to work in an Arduino (1), how do I load only the Slave, if I want the Master to work in the Arduino (2), how do I do it?
I have the problem because when compiling, it does it mostly the innocent. Please could you explain.

budulinek, 2021-12-27:
Hi ayaser,
the code was made for AVR Arduino. There is some AVR-specific codec (see Readme). Maybe, after removing / changing that piece of code, it would be possible to run it on ESP8266 or ESP32. But I do not know, I do not have ESP8266 or ESP32, so I can not test it.  You can try this port:
https://github.com/harihanv/esp32-modbus-gateway

The code is quite long, I have divided it into several ino files. You need all ino files in the folder to compile the code.

ayaser, 2021-12-27:
muchas gracias por tu respuesta.

Adrián Yaser

***@***.***
+56 9 6149 8858 <https://api.whatsapp.com/send?phone=+56961498858>

DVRSOLUTIONS
Servicios y Productos para su Seguridad
<https://api.whatsapp.com/send?phone=+56961498858>

budulinek, 2021-12-27:
BTW, @harihanv was able to run it on ESP32, but his code is obsolete - he did not fork from my repo but created new one  :-( 

Please fork the repo and if you (or anyone else) is successful in porting to ESP32 / ESP8266, feel free to do a pull request. I would be happy to include it.

ayaser, 2021-12-27:
Thank you very much for your answer
The five programs are united in the first one, is correct what I say? those
are not 5 standalone programs.
Therefore, everything would be loaded on the arduino board and then through
the web I would choose which function it would fulfill.
what i said is correct?

harihanv, 2021-12-28:
hi,

 I am sorry for the late reply. I was trying to implement RTU to TCP modbus
gateway. This is more of packet translation
that was the reason i used the code available and ported to esp32.

From your description looks like you need modbus slave / master for your
implementation. We can either port libmodbus
or freemodbus to esp8266.

I will check on nodemcu and update.

Regards,
Harish

ayaser, 2021-12-28:
Sorry, my English is not the best, and I think I got a little lost in the
thread of the conversation, I don't know if all the emails were yours, or
if another person intervened commenting on ESP8266.

First of all, thanks again for your answer.
The project that you wrote is just what I need, that is why I attached a
file in pptx in case you want to modify it or rather correct it.
It is about implementing Modbus, a slave in which I can execute certain
programs, a Master which can give setting values to the Slave, and a
Gateway that allows me to send the parameters to Modbus and deliver its
results to the user, which could be both an operator like the manager.

I am an electronic engineer, I am not an expert in programming, and I am
trying to make a technology integration system, I do not intend to invent
anything, but to develop a system that I can implement to sell and earn
money.
I want to clarify this because I am not an expert in open systems, so I do
not know what is the scope of what I can expect from this type of
contribution, which I see as extraordinary. What I mean is that if your
time had a cost, I am not in a position to pay it now, but if the economic
result were advantageous, I would have no problem sharing it.

Using the translator, I can know if the phrase in English is correct, if
they express what I want to convey to you.
If you have any possibility of being able to do a chat (written), it would
be good if we could talk, logically if you have time.

I am in Chile, with time zone -3.

My cell phone is +56961498858.
Kind regards.

Adrián Yaser

***@***.***
+56 9 6149 8858 <https://api.whatsapp.com/send?phone=+56961498858>

DVRSOLUTIONS
Servicios y Productos para su Seguridad
<https://api.whatsapp.com/send?phone=+56961498858>

harihanv, 2021-12-28:
Request to attach the ppt again. I have taken code from the author who
replied previously; removed ethernet code and added wifi.

If you are looking it to work on esp8266, I need to check on serial port
availability.

Regards,
Harish

ayaser, 2021-12-28:
I apologize. I did not understand his answer.

Adrián Yaser

***@***.***
+56 9 6149 8858 <https://api.whatsapp.com/send?phone=+56961498858>

DVRSOLUTIONS
Servicios y Productos para su Seguridad
<https://api.whatsapp.com/send?phone=+56961498858>

harihanv, 2021-12-28:
Please share the ppt

ayaser, 2021-12-28:
you mean share it on github?

Adrián Yaser

***@***.***
+56 9 6149 8858 <https://api.whatsapp.com/send?phone=+56961498858>

DVRSOLUTIONS
Servicios y Productos para su Seguridad
<https://api.whatsapp.com/send?phone=+56961498858>

ayaser, 2021-12-28:
[What I look for.pptx](https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/files/7784726/What.I.look.for.pptx)

budulinek, 2021-12-28:
@ayaser you are looking for something different. 

My gateway converts between two different modbus protocols (Modbus TCP and Modbus RTU). If you already have a Modbus TCP master (for example a home automation server) and Modbus RTU slave (for example some sensor or energy meter), you can use my gateway to connect them together.

The web UI in my project is only for the configuration of the gateway. You can not use the web UI for sending modbus requests and presenting the results to the user.

ayaser, 2021-12-28:
Thanks for the clarification.
thanks for your answer.
But with your code, the master and slave mode are not also executed, but
they on another card?

budulinek, 2021-12-28:
Modbus master, modbus slave and modbus gateway are different devices. My code is for a modbus gateway.

ayaser, 2021-12-29:
Dear Sir, I am making progress in trying to adapt your code to ESP8266. But I am in the following error, which is in the Interfaces.INO, line 38 "
void startSerial () {mySerial.begin (localConfig.baud, localConfig.serialConfig); ".
Could you guide me and tell me which library this item, object or function belongs to?
Thanks a lot.

ayaser, 2021-12-29:
Dear Sir, I am making progress in trying to adapt your code to
ESP8266. But I am in the following error, which is in the
Interfaces.INO, line 38 "
void startSerial () {mySerial.begin (localConfig.baud,
localConfig.serialConfig); ".
Could you guide me and tell me which library this item, object or
function belongs to?
Thanks a lot.

Adrián Yaser

***@***.***
+56 9 6149 8858 <https://api.whatsapp.com/send?phone=+56961498858>

DVRSOLUTIONS
Servicios y Productos para su Seguridad
<https://api.whatsapp.com/send?phone=+56961498858>

budulinek, 2021-12-29:
I am sorry, I do not have ESP8266. I can not help you more.

harihanv, 2021-12-29:
This belongs to serial class please check arduino esp8266 github for
details. Check the baud rate configuration.

ayaser, 2021-12-29:
thanks for your answer.
In the original file, is this instruction, or function, general or
does it belong to Arduino?

Adrián Yaser

***@***.***
+56 9 6149 8858 <https://api.whatsapp.com/send?phone=+56961498858>

DVRSOLUTIONS
Servicios y Productos para su Seguridad
<https://api.whatsapp.com/send?phone=+56961498858>

harihanv, 2021-12-29:
It belongs to arduino serial handling class specific to esp8266 for serial
uart configuration and communication.

closed 2022-04-20 by budulinek