budulinek/arduino-modbus-rtu-tcp-gateway #10: the last version doesn't work well

dl9msr, 2021-11-15:
Hello,
the last version v3.0 doesn't work well. 
The connection freezes after a few minutes and I have to reset the Arduiono.
Troubleshooting from last issue didn't help
I'm back to version v2.3

Best regards
Michael

budulinek, 2021-11-15:
Hi,
thanks for your report. 

My Arduino is fine at the moment. In version 3.0 I have increased webOutBufferSize from 64 bytes to 128 bytes in order to speed up web UI loading. I thought the step is safe... Could you PLS try to go back to 64 bytes and check whether it helps?

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/ac497050e0fdd2a4397ac155dacb7308a479f9c6/arduino-modbus-rtu-tcp-gateway/05-pages.ino#L26

dl9msr, 2021-11-15:
Hi,

thank you for your prompt reply. I have changed to 64byte.
I'll get back to you tomorrow

Good night.

Michael

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/ac497050e0fdd2a4397ac155dacb7308a479f9c6/arduino-modbus-rtu-tcp-gateway/05-pages.ino#L26

dl9msr, 2021-11-16:
Hello,
I think the problem is my Arduino Mega 2560 on the test board and not your sketch.  I have to take a closer look at that.

I loaded the sketch in the production system with the Arduino Mega 2560 R3 pro mini and it works fine there.

Thanks for your support

budulinek, 2021-11-16:
OK.

closed 2021-11-16 by budulinek