budulinek/arduino-modbus-rtu-tcp-gateway #11: Modbus RTU Slave ID in Modbus TCP/UDP not function

lilisugeng, 2021-11-23:
When using Modbus TCP/UDP the scanned modbus RTU Slave ID may not be stored permanently, causing the master to fail to connect.

budulinek, 2021-11-23:
This is a feature (of all Modbus Gateways), not a bug. It is up to the master to remember the correct slave ID. It is also up to the master to send correctly formated Modbus TCP message (which contains slave ID).

Does the "Scan" button find your RTU slave? It means that RTU slave sent a response to a test command sent by the gateway. It means that your RTU settings are correct. 
Does RTU slave disappear from the "Current Status" page once you connect your master to the gateway? It means that:

- gateway accepted your Modbus TCP message from the master and forwarded it to the RS485 line
- RTU slave did not respond (maybe because the master used wrong slave ID) or responded with an error

PLS check the Modbus TCP message sent by the master.

lilisugeng, 2021-11-23:
The scan button finds the Slave RTU, in this case it finds 0x01 OK, that's also the Slave ID I set on the master.
I'm using a kepware OPC server for this test and I'm using a Schneider PM800 Energy Meter.
The gateway works fine when I use Modbus over TCP, with the same Slave ID setting, but not for Modbus TCP.  Is there anything I should pay more attention to? Thanks

lilisugeng, 2021-11-23:
Sory I think I did wrong in my kepware setup, I use modbus scanner it is working fine also when I use Modbus TCP.
Sorry again for my foolness.
Thanks so much,  the program you have made is realy awsome.  Thank you.

budulinek, 2021-11-23:
Hi, I am glad to hear that everything works :-) Be careful, Modbus TCP and Modbus RTU over TCP are two different things.
Have a nice day!

closed 2021-11-23 by budulinek