budulinek/arduino-modbus-rtu-tcp-gateway #13: Modbus with ENC28J60

ayaser, 2022-01-03:
good afternoon.
I would like to ask if you have used an Arduino card with an ENC28J60 network card for modbus?
This network card, according to the information on the WEB, is used with the UIPEthernet or EtherCard library.
Do you know her
It has had some problems, since during programming, there are some instructions, or objects that are not 100% compatible or transparent.

budulinek, 2022-01-04:
I have very bad experience with ENC28J60. I have tried all available libraries, but could not make it work. 

ENC28J60 is bad chip, because it does not have built-in hardware TCP/IP stack. You have to use additional librraries to get the stack.

Buy W5100,W5200 or W5500 chip. Much better, reliable, with built-in TCP/IP stack.

ayaser, 2022-01-04:
Buena respuesta.
Voy a trabajar con w5500.

closed 2022-01-06 by budulinek