budulinek/arduino-modbus-rtu-tcp-gateway #14: Ethernet was not declared

ayaser, 2022-01-06:
Dear sir, I changed from working with ESP8266 and ENC28J60 to Arduino UNO with W5500, and I get the following error: "Ethernet was not declared ..". That error is indicated in Ethernet.begin, in Ethernet.etc ...
I am reviewing the original program and the Ethernet.h library is perfectly included in it, so I understand that there is no error in it.
Could you please tell me where I should check to fix this error?

Sierra83, 2022-01-06:
Wrong platform
Ask in the Arduino forum

budulinek, 2022-01-06:
@ayaser 

I do not want to be rude..... But this is basic stuff which can not be discussed here. Please Google it, how to setup ethernet on ESP8266.

closed 2022-01-06 by budulinek