budulinek/arduino-modbus-rtu-tcp-gateway #16: Esp32 version

maufc, 2022-05-08:
Hi 
I am trying to make this work on esp32. There are some fork version for esp32 but no one works, at least for esp32 dev board.
Would it be so complicated to adapt this to esp8266/esp32?
I think the memory limits on Arduino borads will be overcome.
The main problem seems to be in ArduinoRS485.h with SERIAL_PORT_HARDWARE variable definition.
Thanks!

hopfi2k, 2022-05-08:
Same here. Maybe https://github.com/plerup/espsoftwareserial/ is the way?
Thanks

budulinek, 2022-05-08:
I do not use ArduinoRS485.h (too big, not needed)!!!! I use Arduino HW serial for the communication with the RS485 module. Since ESP32 also has HW serial ports, modifying my gateway for ESP32 should be easy. I have no experience with ESP32, but I think all you need to do is:

1. replace the Ethernet.h with Wifi.h library and adjust the code accordingly (some functions may be missing in Wifi.h library) 
2. use HW serial port on ESP32 (see https://circuits4you.com/2018/12/31/esp32-hardware-serial2-example/ )
3. find an alternative to these Arduino-specific libraries and functions:

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/5ee77b5b54f74b51e5f2c49246ac2ff60f4e7cb6/arduino-modbus-rtu-tcp-gateway/arduino-modbus-rtu-tcp-gateway.ino#L52

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/5ee77b5b54f74b51e5f2c49246ac2ff60f4e7cb6/arduino-modbus-rtu-tcp-gateway/01-interfaces.ino#L93

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/5ee77b5b54f74b51e5f2c49246ac2ff60f4e7cb6/arduino-modbus-rtu-tcp-gateway/01-interfaces.ino#L151

I use these functions for generating random MAC and for reseting Arduino via Web UI. These are not core functions, you can easily modify the code to use static MAC and drop the function for reseting Arduino....

I know there is a ESP32 port here: https://github.com/harihanv/esp32-modbus-gateway
But it takes the wrong approach by using the ArduinoRS485.h library. As I have already said, forget about ArduinoRS485.h and use HW serial on your ESP32!

Once you have it running, let me know (or do a pull-request) and I would be happy to adjust my code to make it ESP32 compatible.

zivillian, 2022-06-12:
I've started a port, but it's still work in progress. It compiles, the web ui works (sometimes) but there are still many things to to. The current state is available at https://github.com/zivillian/arduino-modbus-rtu-tcp-gateway/tree/esp32

maufc, 2022-06-12:
Looks great!
I'll try this week and let you know!

zivillian, 2022-06-14:
I have to admit, that I gave up and started from scratch with ESP32 libraries. It's far from finished and there's no usable webinterface yet, but the modbus RTU/TCP stuff is already working. If anyone is interested and want's to help - my code is available at https://github.com/zivillian/esp32-modbus-gateway

zivillian, 2022-06-17:
It may not (yet) have all features, but I think I've reached the point where it's usable - if anyone is interested, there is also prebuilt binary to try out.

sefaguntepe, 2023-02-09:
Hello, I want to try the codes. Can you share the file for Arduino IDE?

sefaguntepe, 2023-02-10:
> It may not (yet) have all features, but I think I've reached the point where it's usable - if anyone is interested, there is also prebuilt binary to try out.

Hello, I want to try the codes. Can you share the file for Arduino IDE?

zivillian, 2023-02-10:
@sefaguntepe you can find the code at https://github.com/zivillian/esp32-modbus-gateway There is no file for Arduino IDE, because it's a platformio project. If you don't want to compile it yourself, you can try the binary from the release.

chuc2rk, 2024-08-05:
> I do not use ArduinoRS485.h (too big, not needed)!!!! I use Arduino HW serial for the communication with the RS485 module. Since ESP32 also has HW serial ports, modifying my gateway for ESP32 should [...]

How about edit run on esp32 with W5500?
Thanks!