budulinek/arduino-modbus-rtu-tcp-gateway #17: exceeds available space in board in arduino nano

lilisugeng, 2022-05-18:
Dear Budulinek
I'm failed to build this project with arduino nano and its W5500 ethernet shield (RobotDyn). When I try to verify the program there is a message saying  exceeds available space in board.
But I thing you sucsess with the same hardware.  Please let me know if I did wrong.

Thanks

KjellVerb, 2022-05-21:
Probably due to some library updates... You can comment out the parts that generate a random MAC address and assign a static one, you'll be at 99%.

budulinek, 2022-05-21:
> Probably due to some library updates... You can comment out the parts that generate a random MAC address and assign a static one, you'll be at 99%.

Most probably (did not check yet). On Nano, there were just few bytes left. A library update may add few bytes, causing the issue you mention.

As suggested, try to comment out some code you do not need (random MAC, or some code in the webserver).

KjellVerb, 2022-05-21:
Yeah I started testing with the gateway on a nano yesterday (thanks for sharing your work btw!) and had the same issue, I can confirm that fixed Mac gets you below the threshold.

I'm now trying to get fixed RS485 settings with DHCP enabled because that fits my usecase better. Will fork if it works.

budulinek, 2022-05-21:
DHCP enabled on Nano? As far as I remember, DHCP enabled (dynamic IP) takes a lot of memory... Would require some drastic cuts to the code.

KjellVerb, 2022-05-21:
I'm OK with removing the entire webserver (drastic cuts indeed), I only need a static config with DHCP so that I can assign the IP address from my IOT router.

FYI: removing the webserver and enabling DHCP gets me to 74%, re-enabling random MAC address to 75%

budulinek, 2022-06-04:
Wow, some library got really hungry :-(   A library update (probably StreamLib) eating additional 200 bytes of precious flash....

I did my best to reduce the footprint of the program, I am at the end of my abilities..... 

- Do you guys have any suggestions for further code optimisation?
- Any suggestions for which feature to remove from the default version of the sketch?

budulinek, 2022-06-04:
OK, I have removed ethernet data counter feature on the status page. If you need ethernet data counter, #define ENABLE_EXTRA_DIAG.
Now the program fits on Arduino Nano (tested 4.6.2022, Arduino IDE 1.8.19).

closed 2022-06-04 by budulinek