budulinek/arduino-modbus-rtu-tcp-gateway #19: Compiling Error

Honk815, 2022-09-23:
Hi budulinek,

great work. So I have an issue that I cannot solve. Maybe there is a library missing, but which? Error Code from Arduino IDE is attached. Thank you very much.

Nico
[error.txt](https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/files/9636570/error.txt)

budulinek, 2022-09-25:
It seems you did not properly download the INO files. Go to the main page https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway
Code > Download ZIP

Unzip and open arduino-modbus-rtu-tcp-gateway.ino in Arduino IDE. Arduino IDE will automatically recognize other INO files (you will see them in separate tabs). Compile, upload and have fun!

Honk815, 2022-09-27:
Thank you very much, this solved the problems. I have downloaded the recommendet zip file.This is not ne reduced Code for nano, isn't it? I have got an error while compiling. "Program is too big for arduino nano" something like that. Thanks for help 😊

budulinek, 2022-09-28:
You are right, program too big. There was an update to Ethernet library in July and the library got bigger.

I will try to reduce the size of the code, but it may take me some time.

Honk815, 2022-09-29:
Thank very much! So I'am waiting for :-) 
Like to reed out my PV inverter via modbus :)

budulinek, 2023-01-05:
Fixed!

closed 2023-01-05 by budulinek