budulinek/arduino-modbus-rtu-tcp-gateway #21: "Unknown board"

Hypercubo, 2023-02-27:
Hello, I have a problem with the code does not recognize the board. The model I have is the "atmega328p Xplained mini".

Error is:
287 #error "Unknown board"

Can anyone tellme, how fix it?

My configuration is:
Board: atmega328p Xplained mini (the hardware is same to serie NANO)
Ethernet: ENC28J60 (NANO Ethernet Shield v1.0 "Deek-Robot")
Modbus: MAX485

best Regards.

budulinek, 2023-02-27:
Sorry, ENC28J60 Ethernet chip is not supported. This is a very primitive chip, requires too much code to operate.

Please look for W5100, W5200 or W5500 Ethernet chips.

closed 2023-02-27 by budulinek

Hypercubo, 2023-02-27:
Very appreciate your answer, I'm going to follow your suggestion and I'm going to buy one of the chips you've indicated.
But i think the problem is other, because the error is come when i compile the software.

:(

budulinek, 2023-02-27:
It is not an error, only a warning generated here:

https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/e9b04e0782aadbde82b311b4e275445ba4986fb9/arduino-modbus-rtu-tcp-gateway/01-interfaces.ino#L415

The board you use is unknown, but the code will still compile.

Hypercubo, 2023-02-27:
Ok, ty, i´m try it again when is arrived new ethernet shield.

budulinek, 2023-02-27:
You were right. The #error directive prevented compilation.
Fixed https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/commit/4bdf86b5f0abef4667fef429939ebc155f7168a5