budulinek/arduino-modbus-rtu-tcp-gateway #24: Trouble porting to STM32 Arduino

completehvac, 2023-04-19:
I am having some trouble getting this to work on STM32, have you tried this? I am using W5500 and get odd results when using the library functions such as W5100.readSnPORT(dataAvailable) this function.

If I assign it to a variable like this 
uint8_t SnPORT = W5100.readSnPORT(dataAvailable); and then compare 
if (SnPORT == 80) it works, but

if (W5100.readSnPORT(dataAvailable) == 80) gives incorrect results.

I think it might be a timing issue, but I cannot get it to work! I have tried a basic webserver with the hardware configuration and it works without issue. The only changes from your latest version is that I remove the generateuniqueID and hard code a mac address for testing.

budulinek, 2023-04-19:
Hi,
thanks for testing STM32 + W5500! I have not tried it myself.

My suspicion is wrong data type (cast). As far as I remember, W5100.readSnPORT() is uint16_t. Maybe `if (W5100.readSnPORT(dataAvailable) == 80)` fails because you are comparing different data types? Try to force uint16_t on the constant? Maybe `uint16_t(80)`?

if you use uint16_t in:
`uint16_t SnPORT = W5100.readSnPORT(dataAvailable);` and then compare
`if (SnPORT == 80)`, will it still work??

Also, at the moment I use `byte` and `unsigned int` data types. It would be a good idea to replace them with `uint8_t` and `uint16_t`.

completehvac, 2023-04-19:
Thanks for your reply, I tested and that want the issue, the cause was that I was reading the client port which for some reason reset the SnPORT to 0x00. That part now works, now I have other issues to find.. 
Thanks for your help.

closed 2023-08-22 by budulinek