budulinek/arduino-modbus-rtu-tcp-gateway #32: Porting this code to Arduino OPTAI'm interested in utilizing this project on the Arduino OPTA platform, which is built on the STM32H747XI. Currently, the project is tailored for a different STM32 variant, and I'm reaching out to inquire about the possibility of porting it to Arduino OPTA. I believe this adaptation could greatly benefit users working with this specific platform. If there are any guidelines, steps, or requirements for the porting process to Arduino OPTA (STM32H747XI), please share them, and I'd be willing to assist in the endeavor.

patkepa, 2023-12-12:
Hello!
I'm interested in porting this project to the Arduino OPTA(STM32H747XI)
Are there any suggestions  in regards to replacing the Atmega based code of this project, for STM32?

Thanks!

budulinek, 2024-01-14:
What ethernet chip is used in Arduino OPTA? Which ethernet chip do you want to use with STM32?

The code is tailor-made for the Wiznet W5100/W5200/W5500 ethernet chip. It would be very difficult to "port" the code to different ethernet chip.

papercodeIN, 2024-06-03:
Hi Creator, 

Could you guide me how to implement this in wiznet w5100s-evb-pico (Raspberry Pi Pico) ?

budulinek, 2024-06-03:
> Hi Creator,
> 
> Could you guide me how to implement this in wiznet w5100s-evb-pico (Raspberry Pi Pico) ?

I have no idea on how to port it to RPi. Some functions and libraries used by this project are specific to W5X00 ethernet chips (Ethernet.h library), some functions I use are specific to ATmega chips (random number generator, reset function, also probably the Serial library...).

Disclosure: I am a 100% hobbyist coder with no formal coding education. I made this gateway for fun and for my self-education. A kind of challenge - what else can I fit into a very limited flash and RAM resources of the ATmega chip. I did not and do not have an intention to port this project to bigger and more powerful platforms.

closed 2024-11-05 by budulinek