budulinek/arduino-modbus-rtu-tcp-gateway #35: timeout but slave respond

pmolke, 2024-12-03:
I get an error code 11 when I send a request to the slave.
But with a sniffer I see that the device responds correctly

[image]

[image]

budulinek, 2024-12-03:
Did you try to send your Modbus RTU request through the built-in web interface? What was the result? Ideally, post here the request message, response message and any results you get in the "Modbus Statistics" in web UI.

What was the result you sniffed on the RS485 line? Looking at your picture, the result was 00 e9 00 02 ? That is weird. That is not a correct Modbus message (for a response).

pmolke, 2024-12-04:
[image]

pmolke, 2024-12-06:
No idea?

budulinek, 2024-12-06:
My suspicion is some HW issue rather than a bug in SW. What is your baud rate? Do you use RS485-TTL module with HW flow control? Maybe the response came too fast and the module did not have enough time to switch the flow 🤷

pmolke, 2024-12-06:
Correct. The sniffer is passiv.

I also suspect that the answer comes too quickly. Do you know how quickly the Modbus module needs to switch automatically? So far I've only had modules that had to be switched manually using a PIN.

pmolke, 2024-12-06:
Baud is 9600

budulinek, 2024-12-06:
I did some tests some time ago but I do not remember the exact result. Can also vary depending on the module you use.

I had to introduce Inter-frame delay (far higher than Modbus specs) in order to deal with the switching time of the RS485 modules. The Inter-frame delay (see RTU settings) is a delay between gateway receiving response via RS485 and sending new request (stored in buffer). If the delay was set too low, the request (which was waiting to be sent in the buffer) was sent (via TTL to the module) too early, before the module could switch the flow.

I have 150 ms as default Interframe delay. This value was safe (= high enough) for the RS485 module I use.

budulinek, 2024-12-06:
Your Modbus slave responds in 15 ms. That is very fast response. Maybe even faster than it should be according to Modbus specs. I do not remember what the minimum Interframe delay should be at baud 9600...

pmolke, 2024-12-06:
I have now added a delay of 300 ms to the slave before it responds.

Unfortunately no success.
Unfortunately I don't know why this doesn't work. I need to test another slave device to rule out a problem with my gateway

budulinek, 2024-12-06:
The Interframe delay setting only regulates delay between response received by the gateway and new request sent by the gateway. Changing the setting will not solve your issue.

I think you either have to use RS485-TTL module with manual switching or try to find some (really) fast module with automatic switching.

pmolke, 2024-12-06:
I have now connected another RS485 module and used PIN6.

Now it works as desired.
It's a shame, I would have preferred the other module.

Thanks for the great software.

Would it be time-consuming to add password protection to the web interface?

budulinek, 2024-12-06:
> I need to test another slave device to rule out a problem with my gateway

Good idea. PLS let me know if you have success with other device. 

... and one more idea. Is it possible to lower baud on your slave? Down to let's say 300? Maybe by setting the slave to lower baud you can force the slave to extend the response time (=Interframe delay on the slave). According to Modbus specs, the minimum Interframe delay depends on baud.

pmolke, 2024-12-06:
I still have the problem that sometimes all connections are disconnected and I have to restart the Arduino to get it to work again.

budulinek, 2024-12-06:
Do you have external power source?

pmolke, 2024-12-06:
No, the Board is connected with my Laptop over USB

You mean under-voltage is the problem ?

budulinek, 2024-12-06:
The problem with lost connection is caused by power / reset issues of the Ethernet module. I will update readme with possible solution. Hopefully on Sunday.

budulinek, 2024-12-06:
> No, the Board is connected with my Laptop over USB
> 
> You mean under-voltage is the problem ?

Probably not. USB should provide enough current (with stable voltage) for Arduino, Ethernet module and the RS485 module.

budulinek, 2024-12-15:
> The problem with lost connection is caused by power / reset issues of the Ethernet module. I will update readme with possible solution. Hopefully on Sunday.

Please check https://github.com/budulinek/arduino-modbus-rtu-tcp-gateway/blob/master/Ethernet_SW_Reset.md

And let me know if it works for you.

pj-r, 2025-03-25:
I had the timeouts also with node red and node-red-contrib-modbus but changing the TCP Type from DEFAULT to RTU buffered resolved the issue. Also UDP did work correctly.